GemHunt/CoinSorter #19: Try the caffe Windows port

pkrush, 2015-08-19:
Try a caffe Windows port and add a script and maybe binaries for this usage. 

Even if it's just for classifying with no GPU it would really handy. 

I can see there is a culture pushing against using a caffe Windows port, admirable yes, but in practice for this project it's a bit silly. 

The Unofficial 64-bit Windows port is:
https://github.com/niuzhiheng/caffe

Since VS 2013 is free this might be a great resource:
https://initialneil.wordpress.com/2015/01/11/build-caffe-in-windows-with-visual-studio-2013-cuda-6-5-opencv-2-4-9/

lunzueta, 2015-08-19:
@pkrush, you should check the most updated Windows port of Caffe, which is that of @willyd as mentioned in Caffe's Wiki (https://github.com/BVLC/caffe/wiki/Installation), but not yet in Caffe's website (which can be quite confusing). You'll see that its structure is pretty much the same as that of Linux/OSX (it's quite close from its latest state). As mentioned in Caffe's Wiki, this version also has a cmake-based super builder, which downloads, compiles and links all the required dependencies and solves all those Linux/OSX-to-Windows compatibility issues (such as the inclusion of unistd.h, etc). You just need to use CMake and Visual Studio (in my case I used CMake 3.3.0 and VS2013, and did a 64-bit build, and it worked perfectly). Just for curiosity, I've also tested the super builder with VS2015 and I got some errors that I didn't get with VS2013, so I guess, for now VS2013 should be used.

pkrush, 2015-08-19:
Thanks for the comments, I will look into @willyd ’s work. 

It’s interesting your trying to bring DIGITS to Windows as well. 

Also, thanks for the being the first person to contribute to this repo!

lunzueta, 2015-08-20:
Thanks @pkrush. You can follow up what's being done around Caffe to make Digits work in Windows here: https://github.com/willyd/caffe-builder/issues/6

closed 2015-10-07 by pkrush