GemHunt/CoinSorter #20: Find rare transitional wide/narrow “AM” USA pennies

pkrush, 2015-08-20:
This is a great challenge to proof the worth of the system. It not going to be that that hard once you have a system up and running. You can find these coins by hand sorted enough pennies. Maybe one in 10,000 to 50,000 are like this? I have already built models that pulled out 1982 copper cents from machine sorted copper. It wasn't that accurate, maybe 80-90%, but did not not spend much time trying. 

**Rare transitional wide/narrow “AM” USA pennies:**
* 1982-1992 normally have a Wide AM
* 1993-2008 normally have a Narrow AM

Sort out just the zincs with coin discriminator / Ryedale
Sort out memorials. This leaves you with 1982-2008 pennies
Manually sort some by decade. 82-89, 90-99, 00-08
Get a model working that sorts heads by decade. So split 90-99 apart by memorials from the 80’s and 00’s. 
Manually split some by 90-92 and 93-99
Get a model working that splits the 90’s pennies in two: 90-92 and 93-99. This model should work on both side of the coins. 
Build a model that work on just the memorial side. 
It’s done, then look for narrow AM’s in the 90-92 pile and wide AM’s in the 93-99 and 00-08 piles.

pkrush, 2015-08-24:
If you are interested in trying to sort by date I put over 80,000 labeled penny images out on [GemHunt.com](www.gemhunt.com) to work with. 

I am thinking about doing a contest by posting a larger set of zinc images out on AWS. I'm not fully sure about the rules, maybe something like this:
* Both the obverse and reverse of the coins would be in the same image. 
* Public Set AWS: 5 known wide AM coins in in a set of 50,000 coins. 
* Private Set: 5 known wide AM coins in in a set of 50,000 coins. 

In the private set: 
The code would need to output it's 50 best guesses and need to ID 4 out of the 5 private Wide AM coins. The first person that posts working code gets all the Wide AM coins. One could only try once a week. 

Leave a comment and let me know what you think about this.

leotam, 2015-08-29:
What sort of data augmentation are you using on the training data? Perhaps random rotations would help the accuracy?

pkrush, 2015-08-31:
I bet it would improve results in finding dates, but I have not tried it yet because it would take a long time to train. I am thinking there is a more direct approach. 

So far, I have been using the stock Alexnet network. I right now I am learning more about how the network, the prototxt file, is put together. I need to understand why they work more. Today I am going to try different image sizes and try training just outer ring segments of the coin images. If I can build a model that knows the coin orientation, then I can build a model of just the cropped area of the coin's date and mint mark. 
[image]

pkrush, 2015-09-17:
The big news is I am now able to determine the orientation of the coins.  Having this I was able to crop just the date and mint mark and train on this. I have 1982P, 1982D, and not 1982 pennies manually sorted out and Caffe is accurately able to tell them apart. I am not sure how many you need to train with to tell them apart. I can say it’s less than a 1000. 

Technically how did I get orientation? I manually rotated a 1000 penny images. Then I created 360 images from each one of these, one for each degree. These 360,000 images were then resized to 28x28 and trained on the LeNet network. Being the inner product layer only has 500 weights and I have 360 labels I raised it to 2000. Also quadrupled the weights in the convolution layers to 80 for conv1 and 200 for conv2. The accurately is 46 percent, but the manually training was only accurate to 1-3 degrees. So 46 percent is great. Doing some spot checking when it was off, it was only off by 1-2 degrees most of the time.

pkrush, 2015-12-09:
[Dates are done.](https://www.youtube.com/watch?v=mU1LRAQGpiU)
Mint marks need to be added to the image processing pipeline. Easy.
This is mostly a hardware issue at this point. There needs to be a better build of the hardware to make it worth running to get to this point.

closed 2017-12-06 by pkrush